olentangyfrc/DeepSpace-2019 #9: Create Logging Standards

Connor2573, 2018-12-17:


Connor2573, 2018-12-18:
On December 17th we discussed what we should log for the Turbo Tank Drive. The things we came up with were: distance traveled (in PUs), velocity of the left side (PUs), velocity of the right side (PUs), current draw of the left side, current draw of the right side, and if the turbo button is held. We thought of logging the joystick values in this subsystem but have yet to come to an agreement on where they should be logged.  We attempted in this list to ensure all of these are raw values and not calculated ones. The value closest to being "calculated" is the distance in PU value since we average the value of all four encoders to get the distance. Unsure if it is better to log the distance as an average or as each individual encoder. Next update will be December 19th.

worthp, 2018-12-30:
I have consolidated Logging and Commenting into Coding Standards with sections for each subject, https://github.com/olentangyfrc/DeepSpace-2019/wiki/Coding-Standards . Please start to fill in as you think of stuff

closed 2019-01-10 by worthp