olentangyfrc/DeepSpace-2019 #17: Get the leads to figure out this workflow

worthp, 2018-12-27:


worthp, 2018-12-27:
This is the 3rd project board. We created boards for Fab and Software which seem to be working pretty well so far. With only those two, we did not have a clean place to assign tasks to get the code on the bot and test. We are going to use this project board to manage that process. At least, that's the thought behind this board. This task is to get you leads to talk about the workflow needed to get code on bots in a less than chaotic manner. If this board will not help, or even impede that then we can kill it with whatever workflow you all think is appropriate.

Make sense?

worthp, 2018-12-27:
you are being recruited to help drive. you have good insights that everyone wants to see more of ;-)

Eiim, 2018-12-27:
Given that I don't know anything about this process, I'll still remain quiet for the time being.

worthp, 2018-12-29:
I suspect it will be a work in progress during the first couple of days of build. Just trying to make the decisions conscsious.

closed 2019-01-21 by worthp