olentangyfrc/DeepSpace-2019 #19: Need to Fix DeepSpace-2019 repo.

worthp, 2018-12-27:
The repo is technically a fork of Connor's which is not what we want we need to init DeepSpace-2019 as it's own repo with the code that is currently in the repo.

Let's chat about this before we do anything. Perhaps a joint effort.

NotSketchyBen, 2018-12-27:
We used that method when we changed this repo over to the new visual code structure, which didn't remove the fork dependency. I did some reading and I may be wrong but it seems that we would need to contact Github support so they could change the repo to what they called "normal mode" which would remove the fork dependency.

https://stackoverflow.com/questions/16052477/delete-fork-dependency-of-a-github-repository

worthp, 2018-12-27:
So. I am thinking easiest and safest way is to create a new empty repo. I can re-create projects and open issues. There are only a handful at this point. Then you guys can just a cleant copy of the code - meaning no git info and just init the code base. There is some whacky under the covers stuff that GH does that treats issues/projects separate from code and we would end up creating more noise for ourselves. Let's discuss before break is over. I don't think it will be bad to start fresh.

worthp, 2018-12-27:
so who's gonna contact github support?

NotSketchyBen, 2018-12-27:
Whoever has access to the olentangyfrc account should.

worthp, 2018-12-27:
i was hoping one of you knuckleheads would point that out ... doing it now ;-)

worthp, 2018-12-28:
GitHub support did this for us.

closed 2018-12-28 by worthp