olentangyfrc/DeepSpace-2019 #20: Automate project boards

Eiim, 2018-12-28:
Using automation, project boards can be automated such that closed issues are automatically put into "done," among other functionality. See more info [here](https://help.github.com/articles/about-automation-for-project-boards/).

We should automate all boards to minimize confusion as to that status of issues. For example, #10 is closed but in "To Do," which can cause confusion as to its current state. During build season, things like this could easily be missed, theoretically causing actual problems.

Convinced? Someone with more access to this repo than I should follow the instructions [here](https://help.github.com/articles/configuring-automation-for-project-boards/) for info on converting project boards. It's just a couple clicks with the presets Github gives.

worthp, 2018-12-28:
Yep. We have some boards set up with automation. i will over automation rules. GH has a bit of a gap though. If you change an issue status from the "Issue List" it has a bit of a different behavior from what happens if you change status on the project board. I will take a look at the #10 issue and see what happened.

worthp, 2018-12-28:
Fab board was not set up for automation. I just set some automation rules.

closed 2018-12-28 by worthp

Eiim, 2018-12-28:
Interesting, thanks for the insight. I'm not sure if Github autocloses issues moved to done, and closed issues are supposed to be automoved to done, but it sounds like that second one might not work when closing from the board. Alternatively, automation might not be set up right on that board.

Eiim, 2018-12-28:
Do all boards have insight? Sorry for my last comment, wrote while you were closing.

Eiim, 2018-12-28:
Edit: automation, not insight. Not sure what my brain was thinking.