olentangyfrc/DeepSpace-2019 #21: Fix 'General' Project

Eiim, 2018-12-28:
There are two current issues with the General project. 

Firstly, its description defines it as "For projects that don't fit into 'Software' or 'Fabrication.'" I propose this be changed to "For projects that don't fit anywhere else."

The other issue is that it has no boards. I suggest that the same 3 board setup used elsewhere be added to General. I also suggest using automation, see #20.

worthp, 2018-12-28:
I think Schmidt created this as a first step for general tracking. @olentangyfrcowner do we still want this board? if so, I can set up the board

Eiim, 2018-12-28:
I think this General is good, gives a place to put CAD, website, videography, social media, handbook, flyers, and other assorted things. Althoguh I think CAD/CNC/3D Printing should have its own board.

worthp, 2018-12-28:
just fixed it up

closed 2018-12-28 by worthp

Eiim, 2018-12-28:
Description is still incorrect.