olentangyfrc/DeepSpace-2019 #30: GoPro Time Lapse

olentangyfrcowner, 2019-01-05:
Time Lapse set up from old gopro for Saturday meetings in the proving grounds, or the fab lab.  When the camera is set up and ready for time lapse, please see schmidt for tripod or a mounting location.

gkrueger12, 2019-01-10:
Will drop off GoPro with a clear 64gb card tommorow (1/10/19)

olentangyfrcowner, 2019-01-12:
I would really like this to be a high priority today, time lapse on the black gopro mounted up high in the shop.  See @olentangyfrcowner for location on mounting this camera

priyankaagochiya, 2019-01-12:
Just as a written reference for everyone in case we ever forget how to record in time lapse: if you hold down the shutter (the record button) for three seconds, the camera will record in time lapse

closed 2019-01-30 by olentangyfrcowner