olentangyfrc/DeepSpace-2019 #31: Silver Cabinet Organization Drawer Hacksaws

olentangyfrcowner, 2019-01-05:
This drawer needs to be re-drawn in Solidworks to hold 2 hacksaws, hacksaw replacement blades and metal files.  
Drawer size first needs to be measure width and depth.  The piece will be CNC'd from 1/2" plywood.  Please do not make the piece too deep when the drawer is open, or else it will not insert into the drawer.

olentangyfrcowner, 2019-01-27:
Tyler, please see me about this today, so maybe we can cut the part out on the CNC machine and I can teach you the process, so maybe you can use it in the future.

Eiim, 2019-02-02:
There's two files that have been in and out of this drawer, we need to know which one or both to put in the model. One has a flat spot in the middle and is shorter, the other has a handle at the end and is longer.