olentangyfrc/DeepSpace-2019 #34: VIP Invite for season meetings

olentangyfrcowner, 2019-01-05:
Tweak our previous invite to match season dates and times.  Then see Mr. Chapman about creating mail merge to print the invites.  Logo needs to be black and white logo attached here.  Please use OZone brand
 fonts when typing, or see Schmidt for these fonts as they are new for 2019.  
[image]

mikagika, 2019-01-09:
Any movement on this?

olentangyfrcowner, 2019-01-12:
Kim,
We need the invite finished on Saturday, so that you and Mr. Chapman can run the mail merge.  This need to be done on my teacher computer in lab 1412.  I have the excel file there with the names.

The invite should invite the VIP to any remaining building meetings Friday-Sunday for the remainder of the season.  Think principals, sponsors, etc.

closed 2019-01-14 by sethstobart