olentangyfrc/DeepSpace-2019 #49: Cargo Bay Front (single)

olentangyfrcowner, 2019-01-06:


Eiim, 2019-01-14:
@olentangyfrcowner is this any different than the side cargo bays? If not, can we just shut this issue because we have two built?

PatYoakam, 2019-01-14:
Yesterday I had 5 kids headed up by Patrick working on cutting all the part we need for the 3 cargo bay section that we were asked to start once the docking platforms were completed as well as the two ends were completed. 
If I remember correctly we have about 1/2 the materials cut for the 3 bays.

mikagika, 2019-01-15:
Having at least a half a cargo ship would be ideal, either so code or a human driver can get used to dropping at both the front of the ship and at each of the bays along at least one side.

closed 2019-01-26 by olentangyfrcowner