olentangyfrc/DeepSpace-2019 #64: Team members have little control over issues

Eiim, 2019-01-08:
As a team member with average privileges, I cannot create issues in specific projects, move an issue in a project, assign my self an issue, or assign others my issue. It seems like team members should be able to do at least some of these.

Because I can't assign anyone, @sethstobart.

mrbbking, 2019-01-10:
Mentor, here, having the same trouble.

I can create issues and comment on them, but cannot assign them, and can't add tags.

I tried the GitHub instructions for [making assignments](https://help.github.com/articles/assigning-issues-and-pull-requests-to-other-github-users/) and for [adding tags](https://help.github.com/articles/applying-labels-to-issues-and-pull-requests/)

The problem is that I do not have the checkbox next to items.

cc: @worthp @sethstobart

Eiim, 2019-01-10:
Assigned Seth, Schmidt, and Worth so they will see this. Between the three of them, it seems like they would be able to figure out a system where members and especially mentors have a little more control. In particular, mentors should be able to close issues, I don't see why that would be a problem.

worthp, 2019-01-10:
> I tried the GitHub instructions for making assignments [1] and for
> adding tags [2]
> cc: @worthp [3] @sethstobart [4]
> thread [6].
> Links:

If @sethstobart can't get to this today, I will fix that up tonight. I 
don't have git access at work

worthp, 2019-01-11:
I think I fixed this. Please let me know.

worthp, 2019-01-11:
Mr. King,

I have added you and Mrs. King to the Mentors team which I have set up 
to have Admin rights to the DeepSpace repo. I also started a discussion 
on that team (i have never done that before so we'll what that feature 
is and how it works) with this information.

See if you have the ability to do what you want and let me know if you 
do not and we'll square it out tomorrow.

. I expect that it sends an email

olentangyfrcowner, 2019-01-12:
Currently I do not want everyone having full admin rights.  I think I might have a solution, but need to discuss the further with @worthp @mrbbking

closed 2019-01-21 by worthp