olentangyfrc/DeepSpace-2019 #84: Navigation

NotSketchyBen, 2019-01-09:


NotSketchyBen, 2019-01-09:
Code is competeled and ready for testing. Stored in 84/navigation

worthp, 2019-01-10:
No it's not. ;-) I see some of the confusion. I was not clear in conversations. Consumers of Navigation's api should not know anything about Pigeon (or any device specific anything), similar to hiding Talons in DriveTrain. 
I see you are returning the Pigeon through a method call. Ask yourself what a consumer of Navigation actually wants. It's not the Pigeon. It's the information that the Pigeon collects .... currentHeading, current YAW , etc

So, in shorthand ...  Navigation.getCurrentHeading() { return this.pigeon.getCurrentHeading();}

make sense?

closed 2019-01-13 by Eiim