olentangyfrc/DeepSpace-2019 #85: Decision on tag usage.

worthp, 2019-01-09:
Github has a feature by which we can 'tag' the code base with a logical name and gives us the ability to easily retrieve any tagged code base whenever we want. We all need to be aware of when and under what criteria we want to tag the code in a conscious/strategic way. I know the team used tags last year and i think it was helpful. It would be good if all the leads an mentors around coding develop, agree and understand what that process is and how we can leverage

mrbbking, 2019-01-10:
I suggest a small set of tags, with each branch/checkin/PR including two or three, such as:

1. Feature being added/modified/deleted
2. Which RIO it's meant for (following the branch-by-MAC-address idea)

mikagika, 2019-01-11:
Tag by subsystem(s) affected?

closed 2019-01-28 by worthp