olentangyfrc/DeepSpace-2019 #90: Mec Chassis with open end plate into GrabCAD

olentangyfrcowner, 2019-01-09:
We need a mec chassis with only frame material, wheels, gearbox housing and cim motors.  Needs to have one end plate cut out in the middle from interior side rail to interior side rail.  See Mr. Miller or Schmidt for questions.

topfuel388, 2019-01-09:
Make sure to use the dimensions of the existing mech chassis.  

For cutout at front, make end plates 6" on each side.  I think that is minimum per the rules.  I think that will create a 15.5" opening on the chassis,  Intake arms will eat up another 2in  leaving approx 13.5" opening.

Eiim, 2019-01-09:
The minimum for a bumper-covered gap is 8". Not sure what the requirements
here are.

topfuel388, 2019-01-09:
Does this sketch help?

Eiim, 2019-01-09:
Not when there isn't a sketch attached

topfuel388, 2019-01-09:
It's a hand sketch not a cad sketch

Eiim, 2019-01-09:
Did you intend to attach a picture of it?

topfuel388, 2019-01-09:
I attached it to the email both times.

I will try to get it into git hub now..

topfuel388, 2019-01-09:
[image]

I did it!!!  Had to attach image on PC.  You will have to teach me how to do with phone.  I'm too old for this. 

Miller

Eiim, 2019-01-10:
Per R24, we just need 6 inches from each corner for bumpers, so given a 27.5" bot (see #62), we can have up to a 15.5" opening in the front. That is true to the description you gave in your original comment as well.

topfuel388, 2019-01-10:
Great.  Good job..

Eiim, 2019-01-12:
Got the base (unmodified) chassis, gonna need someone smarter to tell me if there's a difference between front and back though.

topfuel388, 2019-01-12:
Front of chassis is the end with the cutout.

Miller

topfuel388, 2019-01-12:
I looked at the chassis you drew.  I verified the 27 x 32.25 dim.  Looks very good!!

Eiim, 2019-01-18:
Finally, this is done

closed 2019-01-18 by Eiim