olentangyfrc/DeepSpace-2019 #101: Print 1/20th scale field model

Eiim, 2019-01-10:
Use AutomationDirect's 3D-printable field model and parts from 2018's to create a 3D-printed 1/20th scale half field.

Priority list:
1x Rocket Shell_v125
1x Rocket_Shelves_V2
1x Habitat
2x Depot
1x CARGO (will make new model)
1x Rocket Shell_v125
1x Rocket_Shelves_V2
1x CARGO_Rack
1x HatchRack_v2
1x CargoShip_Spacer
13x CARGO (still new model)
1x CARGO_Rack (nice to have)
1x HatchRack_v2 (nice to have)
6x CARGO (still new model) (nice to have)
9x CARGO (still new model) (nice to have)

Eiim, 2019-01-10:
The new cargo model should come from #102

olentangyfrcowner, 2019-01-10:
2 rockets are printed.  Hab platform is printed. Cargo spacer is printed.

Eiim, 2019-01-10:
Great! Assuming you meant both parts of the rocket, here's an updated priority list:

2x Depot
1x CARGO (may make new model)
1x CARGO_Rack
1x HatchRack_v2
13x CARGO (may new model)
1x CARGO_Rack (nice to have)
1x HatchRack_v2 (nice to have)
6x CARGO (may new model) (nice to have)
9x CARGO (may new model) (nice to have)

Eiim, 2019-01-10:
We might want to look into the decals as well, but I don't know much about that.

Eiim, 2019-01-13:
We're in a good place. We _might_ print more further on in the season, but we have everything we want at this point.

closed 2019-01-13 by Eiim