olentangyfrc/DeepSpace-2019 #99: Label RIOs with MAC addresses

mrbbking, 2019-01-10:
Attach a label to the top of each RIO showing the last two bytes of its MAC address.

The Code team is planning to use MAC addresses in the code so that the code itself can know which RIO it's running on. Labeling the RIOs will help the Electric team and the Code team to communicate more clearly.

**Note:** a [MAC address](https://people.richland.edu/dkirby/141macaddress.htm) is six bytes long and shown in hexadecimal like `01:02:03:0A:0B:0C`. The first six bytes identify the manufacturer, so are likely to be the same on all RIOs. Using just the last two should be easier to say aloud (and remember!) and still give us unique labels.

A label for the MAC address above would look like this: `0B:0C`

worthp, 2019-01-12:
(bbk adding)
1D:E9 - RIO 5
04:C6 - RIO 3
D7:4B - RIO 2
F8:3F - RIO 1

mrbbking, 2019-01-21:
Re-reopening for same reason: This was incomplete last I checked, and I have heard no news since then.
A list of RIOs and their MAC addresses must exist before this can be closed.

...where "exist" means "Mr. King has a copy of it."
(If it were on the wiki, then Mr. King would pretty much "have" a copy of it...)

olentangyfrcowner, 2019-01-21:
https://github.com/olentangyfrc/DeepSpace-2019/wiki/Getting-MAC-Address-from-Roborio

mrbbking, 2019-01-21:
Thanks! I edited that page with some clarifications one one suggestion for a safer permissions change. Please review that, @worthp

closed 2019-01-21 by mrbbking