olentangyfrc/DeepSpace-2019 #109: Change SubsystemFactory to Use MAC_ADDRESS

olentangyfrcowner, 2019-01-11:
I set lvuser environment to set MAC_ADDRESS on last year's comp bot and football

mrbbking, 2019-01-12:
Hey @Connor2573 and @NotSketchyBen when you do this, could you please tell Electrics which MAC address goes with which RIO so we can label them? Adding them in a comment to #99 may be easiest.

worthp, 2019-01-12:
I just put @Connor2573 , @NotSketchyBen , and @benschwartz9  names on the assignee list on the electrical issue

mikagika, 2019-01-12:
I have some minor concerns about this. TLDR: needs to be documented because reflash will undo it and I think it would be more symantic to use robot name.

If we're just setting an environment variable, be aware that a reflash will undo that so this needs to be documented somewhere (on the wiki?) such that if somebody needs to reflash a RIO for some reason they can go update it appropriately. 

We possibly could get the MAC address directly instead from an environment variable, but I like the environment variable better because we can switch it easier if we have to switch RIOs out on a bot because the RIO died (or appears to have died). Unlikely, but has happened. 

If we're just setting an environment variable, why not give it a robot name since which subsystems we want to initialize are really based on the robot that the RIO is installed in, not the RIO itself? 

Regardless, if we want to use MAC address, note that there are two MAC addresses on the RIO: USB and Ethernet so we need to determine which one we want to use.

mrbbking, 2019-01-12:
Can we get the MAC address from the OS? I'm thinking something like this (but in Java, obvs)

`mac_addr = exec('$(ifconfig | grep ether | head -1 | cut -d' ' -f 2)')`

On my system, that returns the first address if there are multiple. Unsure if order is reliable.

worthp, 2019-01-12:
> thread [2].
> Links:

yep. that's what i did last night. it's on the football and comp sled

mikagika, 2019-01-12:
Since he said "in Java" I presume he means get it from the ethernet card in the Java code instead of from the environment variable. SO suggests this should be possible, although I've never done it personally: https://stackoverflow.com/questions/37068924/java-printing-ethernet-mac-address

I think though that I prefer the environment variables. A) it's simpler. B) change the rio to a new bot, just put the new bot name/address or whatever and it works, without having to change code. In theory. 

There was talk of getting the identification from a file... could that file maybe be on a flash drive that's inserted into the RIO? Is there a state of something else on the robot that we can read before fully initializing the bot?

closed 2019-01-13 by worthp