olentangyfrc/DeepSpace-2019 #123: Get BYOD access restored for mentors

mikagika, 2019-01-12:


NateRaccoon26, 2019-01-12:
In order for you to gain access you might need a student login, I don’t believe technology gives out keys very quickly

mikagika, 2019-01-12:
We had a mentor's login last year. They apparently expired it before this year.

worthbrandy, 2019-01-13:
I reached out the the Director of Tech last week. He asked me for the login info which I just sent to him so he can fix the login.

mrbbking, 2019-01-21:
Creds I got from @worthbrandy did not work today. Re-opening.

worthp, 2019-01-22:
> Reopened #123 [1].
> thread [3].
> Links:

Next time u are in school, try the same creds again. As it turns out, 
the District WiFi was wonky while we were in school at that time.

mikagika, 2019-01-22:
Did not work previous Friday either.

mrbbking, 2019-01-23:
Sat w/ @worthbrandy this evening to try again all options: the creds do not work.

mrbbking, 2019-01-27:
Please @worthbrandy or @olentangyfrcowner tell me who I can call about this.

I would be able to be more helpful if I could get to GitHub during meetings.

worthp, 2019-01-28:
Is there a reason that people cannot simply plug in to the plethora of ethernet cables/ports that are available in the Eng room ... or simply use the machine at Schmidt's desk?

Mrs. Worth is responsible for more things than most mentors have any idea about. The district has become much more restrictive on access and we will probably not get access anyway.

You know .. I completely rebuilt my linux machine this weekend by plugging in a installing what i needed with the tiniest bit of inconvience of not having wifi. In fact, it was no inconvenience at all. I just sat down at a chair and plugged a cable in. 

Seriously ... is this an issue that we all can't work around with the options that i presented? We are trying to teach kids to overcome obstacles and this thread seems to be the antithesis of that.

I am closing this issue unless someone can tell me it's actually an issue.

@mikagika  @mrbbking  @olentangyowner @worthbrandy @gkrueger12

mikagika, 2019-01-28:
Yes, you can plug directly into the wired network. AFAIK, there's no ethernet drops in the shop or band room though. And why they leave that wide open is a bit of a mystery to me. 

Yes, I can sometimes use my phone as a hotspot, if I happen to have my phone in the right location in that faraday cage of a school. Leaving it sitting on the window sill in the engineering room usually works. Signal in the band room is pretty much non-existent. And probably due to the signal issues, it drains my battery pretty quickly. 

I've tempted to just bring in an old router and plug it in, but that seems to be impolite network behavior at best. People tend to get a little antsy about rouge APs popping up on their network. Seems like the correct behavior to model is to work within the rules to get the access needed.

mrbbking, 2019-01-28:
I though that this BYOD WiFi access was how the school wanted us to do it. I'm trying to follow their rules for their space.

Let me help resolve this. I'd be happy to take this off Mrs. Worth's shoulders completely, if I could get an introduction to whoever on the school's technical team can help.