olentangyfrc/DeepSpace-2019 #132: Clean up wiring for Wonky and Janky

sethstobart, 2019-01-12:
- [ ] 1. Wonky
- [ ] 2. Janky

mrbbking, 2019-01-12:
When we do this, here are the guiding principles: Be neat and orderly. Plan for ease of maintenance and speedy replacement of parts.

Required:

* No exposed conductors.
* All indicator lights must be visible.
* All components (and attached wiring) must be accessible for quick replacement.
* Keep the PCM as far as practical from the radio. I'm told the PCM is RF-noisy and interferes with WiFi reception on the bot.

Try Really Hard To:

* Anchor wires to a stationary point near electrical connections to keep strain off of connections.
* Leave a little slack in wires in case we need to re-terminate the ends or adjust routing slightly.
* Route wires in straight lines.
* Turn in right angles.
* If two or more wires of any color are bundled together, label both ends of each wire with numbers so we can still trace them. (That is, a numbered wire should have the same number on both of its ends.)
* Violate any of these rules before doing anything dumb just to follow them. Think about your options, then be prepared to defend your decision.

Here's a good wiring example that's probably blocked on the school network: https://imgur.com/FBoFD6H

Here's one that's probably accessible: https://simplexsystemcontrols.com/custom-control-panels

mrbbking, 2019-01-21:
Fab is rebuilding these. No progress on wiring today for that reason.