olentangyfrc/DeepSpace-2019 #135: Need to change the location of MAC Address setting on rio

worthp, 2019-01-12:
The script to set MAC Addess needs to be relocated to /home/lvuser/.bash_profile from /home/lvuser/.bashrc. This will correct the the problem we had with MAC_ADDRESS coming back null.

I have already done this on the Football

sethstobart, 2019-01-14:
So we need to regrab the MAC address for each rio (sans football) and relabel physically as well as on hardware?

sethstobart, 2019-01-16:
@Connor2573 @NotSketchyBen. This is something I'd like the new members to be able to do. We have documentation on the process in the wiki but I'd like to see them do it themselves. It's an important part of the code base now and they're gonna have to learn eventually.

closed 2019-01-25 by sethstobart