olentangyfrc/DeepSpace-2019 #149: District Communication Contact

worthbrandy, 2019-01-13:
How do we get Ozone Fundraisers and information out district wide? It is a district wide team.

How to better communicate?

worthbrandy, 2019-01-15:
I spoke with the assistant director. She is looking at how to help us raise money and get more exposure. She is looking into what they can do to promote our fundraisers.

JenniferKing4611, 2019-02-02:
@worthbrandy Any more information from the assistant director.  We have the confirmation from Chipotle for a fundraiser for 2/18 and it would be so great to get this out district wide.  I have the chipotle flyer and can send it on to you or if you want me communications directly, just give me a name.  Thank you.
[224788-flyer.pdf](https://github.com/olentangyfrc/DeepSpace-2019/files/2823827/224788-flyer.pdf)