olentangyfrc/DeepSpace-2019 #178: Design Brackets to connect the kicked back top to the fixed elevator frame.

topfuel388, 2019-01-16:
We need brackets drawn to connect the kicked back top to the fixed elevator frame.  Again we can use the 1360 design as a base.  The top will be made from the same 1 x 2 x 1/16in material as the uprights.  We would like to use the same pop rivets as the Andymark brackets, so you can get the hole sizes from the Andymark model.

Email Miller if you have questions.

topfuel388, 2019-01-16:
bracket material is either 1/8in aluminum or 1/8in carbon fiber

Eiim, 2019-01-16:
My only question is that I have no idea what any of that means. And also why you assigned me as the only person on 5 issues. We have other 5 or 6 other drafters! I'll assign some other people to some of these.

topfuel388, 2019-01-16:
Ethan,

I realize it is difficult to visualize these tasks based on my limited
descriptions.

I would be happy to discuss these assignments with you or whomever u assign
them to.

I assigned them to u because you seemed to be in charge of the drafting
team and I wasn't sure who was who on git hub.  My intention is not to
overload you.  Please share my contact info with whomever u assign.

Feel free to contact me via email or phone at 740-815-2747.  Or
topfuel388@gmail.com.

Miller

Eiim, 2019-01-16:
I believe the whole robotics team is on GitHub. I'll take a look at some
things today.

topfuel388, 2019-01-25:
> [image: 2]
> <https://user-images.githubusercontent.com/46425960/51772598-6448a880-20ba-11e9-9b12-c0a354fe3921.PNG>
> [image: 1]
> <https://user-images.githubusercontent.com/46425960/51772599-6448a880-20ba-11e9-9d10-cb7578e54e4b.PNG>
> In team 1360's CAD model of their robot, you can see the L plates they
> used are really not connected to anything. They didn't model how they
> attached it.

Sounds like we need to figure out how to get it attached.

See me on Friday or Sat to discuss.

Miller