olentangyfrc/DeepSpace-2019 #182: Investigate where we are right now with variable speed controllers.

sethstobart, 2019-01-16:
Do we need to fix any that are broken? Can we come up with a better design. They're using them a lot for prototypes. Might be somewhere we can improve so they aren't bottlenecked by code.

olentangyfrcowner, 2019-01-16:
this would be a great help.  With documentation on how to use it.

mrbbking, 2019-01-21:
Today @NateRaccoon26 fixed one controller and is re-working one of the boards for easier use. We have two of Blane's design working.

Could use something more durable than the 3D-printed case for the electronics, and cleaner layout & wiring on the boards so they're more intuitive and safer to power up.

NateRaccoon26, 2019-01-27:
We now have 3 controllers fixed and working
Blanes controllers have 2 connectors
The third is using a speed controller I found

I am working with fab to make a case for the new one @sethstobart bought and will get that one working likely next week

mrbbking, 2019-01-27:
Yes, @NateRaccoon26 please do write up that usage & repair info. Put it on the wiki here, if you have rights to. If you don't, then write it up somewhere and point me to it. I'll get it onto the wiki.

NateRaccoon26, 2019-01-28:
@mrbbking @olentangyfrcowner Added a link to the repair manual on the wiki, I will add the 3rd type when I get the case from the 3D team and get it working

closed 2019-02-03 by sethstobart