olentangyfrc/DeepSpace-2019 #204: Create 3D printed "posts" for the cargo ship

olentangyfrcowner, 2019-01-19:
@RileyMcGaughey we need to create these posts for the cargo bays.
There are three (3) posts on each side of each HATCH. Each post is ⅝ in. (~2 cm) in diameter and
extends ¾ in. (~2 cm) from the face of the CARGO SHIP. Posts are aligned vertically and positioned such
that the center of the lower post is 1 ft. 1 in. (~33 cm) from the floor and their centers are 5½ in. (~14 cm)
apart. 

[image]

We also need to know the location of the posts.  Once we get a set printed we can give to @priyankaagochiya and @patrickyoakam so they can install them at the correct location.

mrbbking, 2019-01-21:
On the actual field element, these  are just bolts with 5/8" heads and a spacer.

Would it be better (more realistic, more durable) to use bolts instead? See [2019 Field Tour Video: Cargo Ships](https://www.youtube.com/watch?v=QXiK6ChuCrE&t=45) for a clear image of the actual field element: 

cc: @olentangyfrcowner for comment.

Eiim, 2019-01-25:
These have been handed off to @patrickyoakam as of today

RileyMcGaughey, 2019-02-02:
The hatch posts are printed and ready to be put on