olentangyfrc/DeepSpace-2019 #212: Make Models of Electrical Components

mrbbking, 2019-01-21:
I'd like to have a set of electrical part models so we can experiment to find the best arrangement for them on the bots. They should be big enough to work with, but small enough to store neatly.

Overall outside dimensions of each piece are the only important part. We don't need surface features or accurately-rounded corners, so 3D printed with low detail or cut accurately from sheet goods would both work. The height dimension is not critical for this use.

Each part should have its name printed or written neatly on it.

Parts needed:

* RIO
* Power Distribution Panel
* Voltage Regulator Module
* Pneumatics Control Module
* Talons (8 of these)
* Main breaker
* Battery
* Air Compressor (footprint only)
* Air tank (footpring only)

mrbbking, 2019-01-21:
Assigning to @sethstobart and @olentangyfrcowner for help in prioritizing and assigning.

Eiim, 2019-01-21:
Putting under Drafting, although it  depends on how this ends up happening.

Eiim, 2019-01-21:
Note: I think 1/4" is sufficient and will save time and filament

mrbbking, 2019-01-27:
In the set I picked up today, I have

* RIO
* Compressor
* Batter
* PCM
* Talons

I know you were printing another piece today, but I don't remember which one.

Here's what's still missing:

* Power Distribution Panel
* Main Breaker
* Voltage Regulator Module
* Air Tank

Eiim, 2019-02-02:
All models are done and given to Mr. King

closed 2019-02-02 by Eiim