olentangyfrc/DeepSpace-2019 #243: Reachout to Knockout Concepts

DylanMires, 2019-01-26:
They are a local company that builds handheld 3D scanners, I know a former intern to their company and I can acquire contact information. Maybe upon hearing about our program they would be willing to donate one of their 3d scanners, at least for the build season.

DylanMires, 2019-02-01:
@JenniferKing4611 I will be able to get the info on Monday, the person who has the networking with them has all of the contact info on a school computer and with school being out, it hasn't been possible to get it

JenniferKing4611, 2019-02-02:
@DylanMires Thank you.  One thing, Mr. King reminded me that this information is public so best to not post it here.  You can give it to me at a meeting, or Mr. Schmidt can pass it on to me.  Thanks again.

DylanMires, 2019-02-08:
I have gotten the information, where would be the appropriate place to put it?

JenniferKing4611, 2019-02-09:
@DylanMires I will be there this morning.  You can give it to me in person.  Thank you.