olentangyfrc/DeepSpace-2019 #257: Solder PWM connectors to Analog Limit switches

NateRaccoon26, 2019-01-26:
Despite the somewhat contradictory title, this needs to be done and is high priority.

Initially I was tasked to do this, but I just learned I’m not able to be there tomorrow 

Solder Ground to the COM (com is the solver colored one that is on the side)

And then solder signal to I believe LO2 (it’s the gold one closer to the COM)

@sethstobart @mrbbking not able to be there tomorrow I could come in early to do it like 10am
If not I need anyone the knows how to solder to do this first thing, Andy Halter and code is waiting on them

NateRaccoon26, 2019-01-26:
Update: right after I submit the issue I’m told that doors open at 11, I can maybe do it then

mrbbking, 2019-01-27:
I am still sick, but I'm coming in at 11 to try to understand this and help get it done.

sethstobart, 2019-01-27:
@NateRaccoon26 Remember when you make an issue to sort it into the proper project. Otherwise we can't see it when we go to the project board :(

NateRaccoon26, 2019-01-27:
@sethstobart I believe only admins can add tags and move issues

NateRaccoon26, 2019-01-27:
I was able to get them wired, ready for closing

closed 2019-01-29 by sethstobart