olentangyfrc/DeepSpace-2019 #262: PWM cables crimping

olentangyfrcowner, 2019-01-27:
Need to discuss cables.  I'd like to buy a crimper and raw materials to make these in house at custom lengths.  I just want to buy the correct stuff.  

I have a mcmaster link for a crimper.  

https://www.mcmaster.com/catalog/125/827

We can discuss today.

mrbbking, 2019-01-27:
Let's discuss - we have a ton of PWM wire.

In any case, Andymark has kits:

$50 [Crimper and wire ends (no wire)](https://www.andymark.com/products/pwm-connector-kit)(am-2817)

$25 [Wire ends and wire (no crimper)](https://www.andymark.com/products/pwm-cable-making-kit) (am-2868)

$6 [10' PWM wire](https://www.andymark.com/products/22awg-bonded-pwm-cable-black-red-white-10ft)(am-3162)

McMaster-Carr has parts:

$62 [Crimper only](https://www.mcmaster.com/69555k65)(69555k65)

mrbbking, 2019-02-25:
So ... we *used* to have a ton of PWM wire...

I saw that @mikagika picked up an inexpensive crimper and a bunch of ends. I like the idea of making cables of the right length for the competition bot (as something to do at the very last minute, or even onsite at a Regional).

I imagine measuring the necessary lengths once all the details are known, making the cables as we can (away from the bot), then quickly swapping them in for the generic ones.

Would need to buy a big spool of wire to make this effective. Ten-foot lengths (as Andymark has) will probably mean a lot of unnecessary waste.

Hey, @mikagika - can you post a link to the stuff you bought? I'd like to cost this out.

mikagika, 2019-02-25:
Crimper: https://www.amazon.com/gp/product/B01N4L8QMW/ref=ppx_yo_dt_b_asin_title_o01_s01 

Connectors, but I don't know that this is really the best deal for us as there's a lot of parts that are larger than we generally need. You can probably buy bags of just female ends and 1x3 and/or 1x1 cheaper. Nonetheless: https://www.amazon.com/gp/product/B0714B4D2Q/ref=ppx_yo_dt_b_asin_title_o01_s00

Wire, because I needed 5-conductor for Pixy and because then it works well for RGBW LED projects. This seemed like a reasonable Amazon price for 5-conductor, but the insulation on the edge of what works with the connectors. The wire is only 22awg but insulation feels heavier. So the good news is that it should be durable. Again, if we normally only need 3-conductor, there may be better options that are 3-conductor.
 https://www.amazon.com/gp/product/B06XRWK84Q/ref=ppx_yo_dt_b_asin_title_o01_s00

topfuel388, 2019-02-25:
Looks like the Red,black,white PWM wire is referred to as servo wire. Its
#22.

https://www.amazon.com/gp/product/B06XRWK84Q/ref=ppx_yo_dt_b_asin_title_o01_s00

As for the pins and housings.  I bought a bunch of them from Digi-Key a few
years ago.  I will try to look at old orders to see if I can find the Part
Numbers.

Miller

topfuel388, 2019-02-25:
We order Digi-key about once a week or so.

If you want me to, I can order a few of these to verify they are the
correct ones.

3-pin Housings - WM2801-ND
Male Pins  WM2517-ND
Female Pins  WM2510-ND

Miller