olentangyfrc/DeepSpace-2019 #270: Figure out stringing and join gaps with Major Tom

Eiim, 2019-01-27:
Major Tom's last prints have had two issues: stringing and gaps on top surfaces next to sides. Pictures are attached.
[image]
[image]
Look into what might cause these and how we can fix them.

morganhowe, 2019-01-27:
Printer is printing, can't do anything right now.
https://www.simplify3d.com/support/print-quality-troubleshooting/#holes-and-gaps-in-floor-corners
https://www.simplify3d.com/support/print-quality-troubleshooting/#stringing-or-oozing

Eiim, 2019-01-31:
Today's re-print of the pot holder confirms that the gaps have been fixed but the stringing remains, and is possibly worse.