olentangyfrc/DeepSpace-2019 #293: Creating a list of events we want to record

sethstobart, 2019-01-28:


Eiim, 2019-01-28:
WIP spreadsheet:

Also as a Google sheet, if anyone wants access to it let me know.
[2019 Scouting objectives.xlsx](https://github.com/olentangyfrc/DeepSpace-2019/files/2804943/2019.Scouting.objectives.xlsx)

Eiim, 2019-01-28:
Updated with obvious fixes:
[2019 Scouting objectives (1).xlsx](https://github.com/olentangyfrc/DeepSpace-2019/files/2804966/2019.Scouting.objectives.1.xlsx)

Eiim, 2019-01-30:
Current list of all possible (?) events
[2019 Scouting objectives.xlsx](https://github.com/olentangyfrc/DeepSpace-2019/files/2810581/2019.Scouting.objectives.xlsx)

Eiim, 2019-01-31:
Added some info on TheBlueAlliance, for insight as to what could be automated. I might have thought of a few things since the last upload as well. Can't come up with anything else, waiting on feedback from some mentors or other students in case I'm missing anything. Please not that this is supposed to be _everything_ we _could_ track that happens during a match, and we'll narrow that down and combine some things to hopefully not much more than 10 things.
[2019 Scouting objectives (2).xlsx](https://github.com/olentangyfrc/DeepSpace-2019/files/2818773/2019.Scouting.objectives.2.xlsx)