olentangyfrc/DeepSpace-2019 #398: Review custom scouting form functionality

Eiim, 2019-02-12:
I've linked to an example form. Consider this a "tech demo." It's designed for mobile, but usable on desktop as well. Features:

- ? button text to Match Number input tells you what matches you are assigned to
- ? button next to Team Number input tells you what team you're scouting
- Remembers attendance number upon new response
- Increments match number upon response
- Doesn't require page reload for response submissions or new responses, important at events where bandwidth is limited
- Features numerical and checkbox input, as well as a custom input for scoring hatch panel/cargo placements. More input types can of course be added.
- The code isn't beautiful, but at least it makes _some_ sense

Feedback would be appreciated, especially if you can test it on your mobile devices to check if anything is off. How it looks in Chrome is the "canonical" version, any differences in Safari, Firefox, or Opera mobile would be great to know about. I haven't been able to test all of these, especially because Chrome is much nicer to test developments in.

Please note that this is _not_ intended to be a complete form. Of course, main period, end game, and other inputs will be added or changed, per #293.

Again, any feedback on the look and feel or suggestions for custom inputs would be appreciated. The website will be updated semi-regularly with additional functionality, so check back occasionally. I've assigned a few people who have expressed interest or who I think would be valuable in this process. Of course, all are welcome to take a look and give feedback

(https://script.google.com/a/macros/we3chapmans.com/s/AKfycbxCdeN3t-Ni8DkfZ5ga8BeWKD9xSl37PLOCUJeGOcrRrmUrAQDd/exec)

JacobBraniger, 2019-02-22:
It's really good so far. My one complaint is how long the return time is for the scouting information. If there is any way at all to speed that up, please do or add a loading icon or block the button until the report is received or something. I garuntee there will be someone who manages to click it so many times they end up missing an entire match because they only saw those reports for like 5 minutes straight.

Eiim, 2019-02-23:
Can you elaborate which buttons took long to respond?

Most likely, you caught the server when it was down and it had to restart
and reconnect, which takes about 10-20 seconds. This should happen only
once or twice per day at events, as scouts will be keeping the server up.
Process times for functions are all around 1 second or less.

Adding a loading image for the form submission is doable, although the team
number and match requests, which are more prone to long load times because
they are generally executed earlier.

Eiim, 2019-02-28:
New version, new buttons, new bugfixes! Specifically, support buttons were added (we'll discuss Sat.) and Yellow/Red card indicators were fixed to reset properly upon a new submission. If you find any issues, let me know